unitreerobotics/unitree_sdk2_python #13: Initialization error when using with ROS 2

destroy314, 2024-10-19:
When calling `rclpy.init()` and `ChannelFactoryInitialize(0)` in the same program, depending on the order of execution, either the error `[rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Precondition Not Met` or `[ChannelFactory] create domain error. msg: Occurred upon initialisation of a cyclonedds.domain.Domain` will occur. The root cause is that both ROS 2 and the SDK attempt to initialize a DDS domain with ID 0 by default. This can be resolved by modifying the domain used by ROS 2 via `export ROS_DOMAIN_ID=1`.

closed 2024-10-19 by destroy314

vanbacvb, 2025-01-13:
when i set export ROS_DOMAIN_ID=1, i can't see G1 topics anymore. what should i do?

destroy314, 2025-01-13:
> when i set export ROS_DOMAIN_ID=1, i can't see G1 topics anymore. what should i do?

I don't know if there is a way to interact with both the unitree sdk2 and ROS at the same time, maybe there is a way to modify the code of one to use a different domain id, or use another program to forward messages.

theanhnguyen3112, 2025-06-24:
Have you resolved it?

blogdefotsec, 2025-07-25:
Run one part of the prog in lxd.