unitreerobotics/unitree_sdk2_python #19: Error Running High-Level Control Example

BeNavon, 2024-11-13:
I am trying to run the high-level control example `go2_sport_client.py` using the `unitree_mujoco` simulation and specifying the network interface `lo`. When choosing several test options, I encountered the following error message:
```
~/Unitree/unitree_sdk2_python/example/go2/high_level$ python3 go2_sport_client.py lo
WARNING: Please ensure there are no obstacles around the robot while running this example.
Press Enter to continue...
Enter id or name: 
0
Test: damp, test_id: 0
Updated Test Option: Name = damp, ID = 0

[ClientStub] send request error. id: 252906806156372
```
Could you please provide guidance on resolving this issue?
Thank you!

sdhou, 2024-11-22:
你好 我遇到了同樣的問題 請問你現在解決了嗎？

BeNavon, 2024-11-22:
> Hello, I had the same problem. Have you solved it now?

Not yet. I also opened a work order on the Unitree technical support, but their response wasn’t helpful.