unitreerobotics/unitree_sdk2_python #22: G1 high level api only works for damping and standing mode

YandongJi, 2024-11-30:
Hi I tried running `python3 ./example/g1/high_level/g1_loco_client_example.py eth0` but it only works for standing and damping mode, could you check whether this works on your end? Thanks

YandongJi, 2024-11-30:
add self.Start() before this line https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/unitree_sdk2py/g1/loco/g1_loco_client.py#L96

closed 2024-11-30 by YandongJi

Rusteam, 2024-12-10:
> add self.Start() before this line
> 
> https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/unitree_sdk2py/g1/loco/g1_loco_client.py#L96

did you get the `Squat()` and `StandUp()` commands running fine?
For me the robot goes into unbalanced and falls when these methods are called.