unitreerobotics/unitree_sdk2_python #24: G1 examples use unstable stand/sit

Rusteam, 2024-12-10:
Tried running G1 examples for basic operations like standing, squating, but none of them is balanced. The robot is unstable when getting up or squating. How to run these examples properly so that they perform similar to the gamepad buttons?

```
        if test_option.id == 0:
            sport_client.Damp()
        elif test_option.id == 1:
            sport_client.StandUp()
        elif test_option.id == 2:
            sport_client.Sit()
        elif test_option.id == 3:
            sport_client.Move(0.3, 0, 0)
        elif test_option.id == 4:
            sport_client.Move(0, 0.3, 0)
        elif test_option.id == 5:
            sport_client.Move(0, 0, 0.3)
        elif test_option.id == 6:
            sport_client.LowStand()
        elif test_option.id == 7:
            sport_client.HighStand()
        elif test_option.id == 8:
            sport_client.ZeroTorque()
        elif test_option.id == 9:
            sport_client.BalanceStand(0)
        elif test_option.id == 10:
            sport_client.Start()
        elif test_option.id == 11:
            sport_client.Squat()
```

ting-zheng, 2025-01-05:
hello,I'm having the same problem as you ,do you save it?

Rusteam, 2025-01-05:
Waiting for the developers to add support for R1+X mode in this sdk

xkhainguyen, 2025-01-17:
@Rusteam 
Hi, are you able to get any of those scripts working in Mujoco simulation?

Rusteam, 2025-01-17:
> @Rusteam 
> Hi, are you able to get any of those scripts working in Mujoco simulation?

I have not tried that.

Fatyhich, 2025-02-09:
> [@Rusteam](https://github.com/Rusteam) Hi, are you able to get any of those scripts working in Mujoco simulation?

Hello, I tried. It doesn't works. Seems that there is no connection with robot with simulator. Because [here in code](https://github.com/unitreerobotics/unitree_sdk2_python/blob/3af1610228977f449e22c8fba868bb8516d41ec3/example/g1/low_level/g1_low_level_example.py#L95) I always receive `None`. So I think it means there is no communication between simulator and sdk in this example. Could you solve it ?