unitreerobotics/unitree_sdk2_python #28: Unable to Get Low-Level State of G1 Robot with unitree_sdk2_python

vkgo, 2024-12-25:
**Description:**

I am unable to retrieve the low-level state of my G1 robot using the `unitree_sdk2_python` library. I have tried both `read_lowstate.py` (from the 'unitree_sdk2py/test/lowlevel' folder) and `g1_ankle_swing_example.py` (from the 'example/g1/low_level' folder), but neither approach is working.

**Environment:**

*   **Robot:** Unitree G1-EDU
*   **SDK:** unitree_sdk2_python (installed from source using `pip3 install -e .`)
*   **Python:** 3.8
*   **OS:** Ubuntu 22.04 LTS

**My attempt:**

1. **Test 1: Using `read_lowstate.py`:**
    *   Run: `python3 ./example/go2/low_level/read_lowstate.py <network_interface>` (e.g., `python3 ./example/go2/low_level/read_lowstate.py enx04e7a1b6478`)
    *   **Expected Result:** Continuous output of the low-level state data.
    *   **Actual Result:** No output is displayed on the console.

2. **Test 2: Using `g1_ankle_swing_example.py`:**
    *   Run: `python3 ./example/g1/low_level/g1_ankle_swing_example.py <network_interface>` (e.g., `python3 ./example/g1/low_level/g1_ankle_swing_example.py enx04e7a1b6478`)
    *   **Expected Result:** The robot's ankle joints should move, and the low-level state should be printed to the console.
    *   **Actual Result:** The following error occurs:

    ```
    (rl_gym) weijin@ps:~/source/Humanoid/unitree_sdk2_python$ python3 ./example/g1/low_level/g1_ankle_swing_example.py enxe04e7a1b6478
    WARNING: Please ensure there are no obstacles around the robot while running this example.
    Press Enter to continue...
    [ClientStub] send request error. id: 28196582953518
    Traceback (most recent call last):
      File "./example/g1/low_level/g1_ankle_swing_example.py", line 194, in <module>
        custom.Init()
      File "./example/g1/low_level/g1_ankle_swing_example.py", line 95, in Init
        while result['name']:
    TypeError: 'NoneType' object is not subscriptable
    ```

**Request:**

Please investigate why the low-level state cannot be retrieved on the G1 robot using `unitree_sdk2_python`. Any assistance in resolving this issue would be greatly appreciated.

closed 2024-12-27 by vkgo

Fatyhich, 2025-01-25:
dear @vkgo , hello ! 

I have same error message with: `while result ['name']`. if you found solution please tell me how to deal with it ?

change-every, 2025-05-09:
Did you find the solution?