unitreerobotics/unitree_sdk2_python #36: import 錯誤

Baominchao, 2025-01-23:
Traceback (most recent call last):
  File "/home/unitree/Documents/HJ/unitree_sdk2_python/./example/front_camera/camera_opencv.py", line 1, in <module>
    from unitree_sdk2py.core.channel import ChannelFactoryInitialize
  File "/home/unitree/.local/lib/python3.10/site-packages/unitree_sdk2py/__init__.py", line 1, in <module>
    from . import idl, utils, core, rpc, go2, b2
ImportError: cannot import name 'b2' from partially initialized module 'unitree_sdk2py' (most likely due to a circular import) (/home/unitree/.local/lib/python3.10/site-packages/unitree_sdk2py/__init__.py)
按照官方教程安裝後，無法執行實例，python3 ./example/helloworld/publisher.py
報錯如下：
Traceback (most recent call last):
  File "/home/unitree/Documents/HJ/unitree_sdk2_python/./example/helloworld/publisher.py", line 3, in <module>
    from unitree_sdk2py.core.channel import ChannelPublisher, ChannelFactoryInitialize
  File "/home/unitree/.local/lib/python3.10/site-packages/unitree_sdk2py/__init__.py", line 1, in <module>
    from . import idl, utils, core, rpc, go2, b2
ImportError: cannot import name 'b2' from partially initialized module 'unitree_sdk2py' (most likely due to a circular import) (/home/unitree/.local/lib/python3.10/site-packages/unitree_sdk2py/__init__.py)

sunsleaf, 2025-02-02:
遇到同样的问题，请问可以修复一下吗？谢谢！

ahaxxx, 2025-03-08:
我遇到了相同的问题，同时使用命令`python3 -c "import unitree_sdk2py"`也会产生相同的报错

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/wanren/.local/lib/python3.10/site-packages/unitree_sdk2py/__init__.py", line 1, in <module>
    from . import idl, utils, core, rpc, go2, b2
ImportError: cannot import name 'b2' from partially initialized module 'unitree_sdk2py' (most likely due to a circular import) (/home/wanren/.local/lib/python3.10/site-packages/unitree_sdk2py/__init__.py)

我的测试环境是
Ubuntu 22.04 Desktop
按照官方教程安装了unitree_sdk2和Mujoco

OldDriver0907, 2025-03-22:
Hi, you may clone the github repo, not using pip install:
```
git clone https://github.com/unitreerobotics/unitree_sdk2_python.git
cd unitree_sdk2_python
pip3 install -e .
```
This can solve the problem.

chaitanya1chawla, 2025-03-22:
Thanks! This solved it!

> Hi, you may install the unitree_sdk2 package from github repo, not directly using pip install:
> 
> ```
> git clone https://github.com/unitreerobotics/unitree_sdk2_python.git
> cd unitree_sdk2_python
> pip3 install -e .
> ```
> 
> This can solve the problem.