unitreerobotics/unitree_sdk2_python #39: Can't run g1 high_level example after running low_level examples

tsh9l, 2025-02-10:
As title, I can run g1 high_level example just fine when i first boot up G1.
However if i run low_level examples, i cannot run high_level examples anymore, below are some errors when i ran g1_loco_client_example.py , this also happens with the unitree_sdk2 (c++ version)

```
python g1_loco_client_example.py enp0s31f6
WARNING: Please ensure there are no obstacles around the robot while running this example.
Press Enter to continue...
Input "list" to list all test option ...
Enter id or name: 
list
damp, id: 0
stand_up, id: 1
sit, id: 2
move forward, id: 3
move lateral, id: 4
move rotate, id: 5
low stand, id: 6
high stand, id: 7
zero torque, id: 8
wave hand1, id: 9
wave hand2, id: 10
shake hand, id: 11
Updated Test Option: Name = None, ID = None
Enter id or name: 
1    
Test: stand_up, test_id: 1
Updated Test Option: Name = stand_up, ID = 1
[ClientStub] send request error. id: 26786514315927
Enter id or name: 
8
Test: zero torque, test_id: 8
Updated Test Option: Name = zero torque, ID = 8
[ClientStub] send request error. id: 26815997412987
Enter id or name: 
```

PocaBBQ, 2025-02-11:
Look for these lines in g1_low_level examples.
These line make G1 change into Release mode, which mean you can't call API to perform high level action anymore.    

>     def Init(self):
>         self.msc = MotionSwitcherClient()
>         self.msc.SetTimeout(5.0)
>         self.msc.Init()
> 
>         status, result = self.msc.CheckMode()
>         while result['name']:
>             self.msc.ReleaseMode()
>             status, result = self.msc.CheckMode()
>             time.sleep(1)

Try to change back to 'ai' mode, you can do can High level API again
Btw, this is my experience, not from official answer of Unitree. Please do it with caution.

kumar-abhina, 2025-04-23:
How to change the robot mode back to ai mode