unitreerobotics/unitree_sdk2_python #42: Unable to Trigger "WaveHand" and "ShakeHand" in Python, C++, and ROS2

tsh9l, 2025-02-12:
I am unable to trigger the "WaveHand" and "ShakeHand" in Python, C++, or ROS2 scripts. However, these actions work fine with the remote control.

- When I attempt to trigger the arm tasks via Python and C++ scripts, nothing happens.
- When I attempt to trigger the arm tasks with ROS2 using the following command:
```
ros2 topic pub /api/loco/request unitree_api/msg/Request '{
    "header": {
        "identity": {"id": 1, "api_id": 7106},
        "lease": {"id": 0},
        "policy": {"priority": 1, "noreply": false}
    },
    "parameter": {"data": 1},
    "binary": []
}'
```
And then subscribe to /api/loco/response, I receive the following response with an error code 3203 (API not implemented on server):
```
header:
  identity:
    id: 1
    api_id: 7106
status:
  code: 3203
data: ''
binary: []
---
```

The actions can be triggered via remote control.

jliphard, 2025-02-13:
This is the design behavior of the system. 

* When the G1 boots, it is in damp mode. 
* When you turn it on via the remote, "L1+A" and "L1+UP", the system is ready to move, using the `ai sport client`
* The system is now responsive to SDK commands

When you transition to **develop mode**, via the hand controller, "L2+R2", and "L2+A", and "L2+B", this turns off the `ai sport client`, and the system does not respond to rpm mediated SDK commands. You can renewable the `ai sport client` via the app. 

As expected, transitioning to **develop mode** Dows not affect the LED control or the Chinese-only text to speech, such the audio client is not disabled by transition to **develop mode**

I

Rusteam, 2025-02-26:
@tsh9l  have you updated the robot firmware to `1.3.0`?

Fricodelco, 2025-03-01:
> * The system is now responsive to manual controller and SDK commands
Seems wrong, we want to tried to operate hands with stabilization enabled with python API, seems impossible with this firmware :( After SDK connection ai sport client died, any tips?

tsh9l, 2025-04-14:
After updating to v1.3.0 it solves the issue

closed 2025-04-14 by tsh9l