unitreerobotics/unitree_sdk2_python #43: Instructions to control G1 movement via SDK - do not enable *develop* mode

jliphard, 2025-02-13:
- When the G1 boots, it is in `damp` mode.
- When you turn it on via the remote, "L1+A" and "L1+UP", the system is ready to move, using the **ai sport client**.
- The system is now responsive to manual controller and SDK commands

When you transition to **develop mode** via the hand controller, "L2+R2", "L2+A", and "L2+B", **this turns off the ai sport client**, and the system therefore does not respond to rpc mediated SDK commands. 

You can re-enable the ai sport client via the app - DEVICE:SERVICE STATUS:ai_sport/8.0.3.0. This is greyed out in develop mode.

As expected, transitioning to **develop** mode does not affect the LED control or the Chinese-only text to speech, such the AudioClient is not disabled by transition to **develop** mode.

TLDR: Transitioning to **develop** mode turns off the `ai sport client`, turning off high level movement commands such as `Shake Hands` and `Move`.

Fricodelco, 2025-03-01:
got same issue! any tips ?
we want to teleoperate hands with stabilization enabled, seems impossible with this firmware :(

blogdefotsec, 2025-06-03:
Note:

Under the current system version, once the G1 is turned on, the built-in motion control program will automatically start, even if you do not operate the remote control. The program periodically sends commands with a speed of 0. However, if you use the SDK in this state, you may cause conflicting instructions and thus cause G1 to jitter.

Therefore, when you need to use the SDK for development debugging, make sure that G1 is in debug mode to stop the motion control program from sending instructions, so as to avoid potential instruction conflict issues. You can press L2+A to confirm that you are in debug mode.

If the behavior after pressing L2 + A does not match with the teaching video, you can press L2 + R2 several times to ensure entering the debugging mode.

You can see this at https://support.unitree.com/home/en/G1_developer/quick_start.
Lol. Debug mode is desigened for no high level control.