unitreerobotics/unitree_sdk2_python #47: b2 robot has no __init__.py

psan3333, 2025-03-18:
Hello! I was trying to install unitree_sdk2py with "pip install unitree_sdk2py" and I've got an error saying that there was no b2 module. So I fixed it by adding __init__.py into source build and it worked.

anahrendra, 2025-04-03:
hi, @psan3333 do you have a B2 robot with you? I was wondering how to connect to the robot's internal PC. I tried to access unitree@192.168.123.164 with `123` as the password, but it does not work haha. Looking forward if you have any experience with using B2 for development. Thanks!

psan3333, 2025-04-04:
Hello, @anahrendra! I was installing this lib globally on my computer for go2 robot manipulations and eventually succedeed, but **I've never used B2 robot** - it's just this bug was revolved around __init__.py. I can share with you fixed version of this lib.

Also note: there are some .so dynamic libs will be missing in the package, so you might need to move missing libs from this .zip into library package inside python site-packages folder of unitree_sdk2_python (console will show you missing libraries).

[unitree_sdk2_python.zip](https://github.com/user-attachments/files/19602612/unitree_sdk2_python.zip)