unitreerobotics/unitree_sdk2_python #53: Problems with cyclonedds while using NVIDIA Jetson

Orile14, 2025-03-27:
Hey,

I've been working with the SDK for a while, and recently, I tried installing it on an NVIDIA Jetson. However, I keep encountering this error:

Could not locate cyclonedds. Try to set CYCLONEDDS_HOME or CMAKE_PREFIX_PATH

I followed the fix mentioned in the README, but it didn’t work. I also tried installing it through different methods, including the one linked in the README, but the issue persists.

Could this be related to the Jetson’s aarch64 architecture? I'd really appreciate any help you can provide.

Thanks!

Taek-16, 2025-04-11:
What ended up working was manually building and installing the CycloneDDS C library. I handled it inside a Dockerfile like this:
```
# Manually build and install CycloneDDS C library
WORKDIR /opt
RUN git clone --branch 0.10.2 https://github.com/eclipse-cyclonedds/cyclonedds.git \
	&& cd cyclonedds \
	&& mkdir build && cd build \
	&& cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local \
	&& make -j$(nproc) \
	&& make install

# Set environment variable so CMake can locate the installed CycloneDDS
ENV CMAKE_PREFIX_PATH=/usr/local:$CMAKE_PREFIX_PATH
```
After doing this, the SDK picked up CycloneDDS without any issues.

Hope that helps!

Kevinagreenhand, 2025-07-11:
Hello,

Have you found the solution to this problem yet? Unluckily, we also faced this problem while using NVIDIA Jeston and we have tried almost all the methods we could find in the web(including the method that Taek-16 provides here) but finally failed.

Have you found out why this problem happen in NVIDIA Jeston rather than our personal PCs and how to solve this problem? If so, could you please kindly tell us how to let the SDK pick up cyslonedds correctly? We are looking forward to your help!

Thanks a lot!

kvnyng, 2025-07-23:
> What ended up working was manually building and installing the CycloneDDS C library. I handled it inside a Dockerfile like this:
>
> ```
> # Manually build and install CycloneDDS C library
> WORKDIR /opt [...]

This worked for me as well.

Kevinagreenhand, 2025-07-27:
There's another rather simple solution to this problem. We upgraded the pip pakage for python 3.8,  installed onnx ( we do this just because pip warns that we need it, though it may not have made sense in the seccess ), and installed every required packges one by one by using pip again and again, and finally we could install cyclonedds without any error. 

Hope this may help.