unitreerobotics/unitree_sdk2_python #54: [ClientStub] send request error. id: 604196651653

chenjiafu-George, 2025-03-29:
sudo python3 ./g1_loco_client_example.py ens33
WARNING: Please ensure there are no obstacles around the robot while running this example.
Press Enter to continue...
Input "list" to list all test option ...
Enter id or name: 
10
Test: wave hand2, test_id: 10
Updated Test Option: Name = wave hand2, ID = 10
[ClientStub] send request error. id: 604196651653
Enter id or name: 

[image]

我在运行g1机器人的时候，遇到了这个问题如何解决？

kumar-abhina, 2025-04-24:
did you solved the probelm?

andyuan997, 2025-04-24:
The same issue

thomasschichl, 2025-05-15:
Did someone find a solution for this issue?- I have been trying to make the high_level commands run for two days now without success.

William-wAng618, 2025-06-10:
Same here, no success when run any scripts under /g1/high_level.

However, /g1/low_level/g1_low_level_example.py is work

EthanTong1, 2025-06-12:
请问问题是怎么解决的?

Indro-ChrisMoffatt, 2025-08-27:
I got it working. In the unitree app, I navigated to "service status" in the settings and saw that the ai_sport version was 8.0.3.0. I found some documentation somewhere that the LOCO_SERVICE_NAME in unitree_sdk2_python -> unitree_sdk2 -> g1 -> loco - g1_loco_api.py must be "sport" for ai_sport versions >= 8.2.0.0. So I just changed that variable back to "loco" and it worked.

[image]