unitreerobotics/unitree_sdk2_python #62: How to use G1 high level loco client in mujoco simulation

wuxiaoqiang12, 2025-04-23:
when I run the simulation and call `python g1/high_level/g1_loco_client_example.py 'lo'` error occur,
```shell
WARNING: Please ensure there are no obstacles around the robot while running this example.
Press Enter to continue...
1745407121.143813 [1]     python: selected interface "lo" is not multicast-capable: disabling multicast
Input "list" to list all test option ...
Enter id or name:
list
damp, id: 0
Squat2StandUp, id: 1
StandUp2Squat, id: 2
move forward, id: 3
move lateral, id: 4
move rotate, id: 5
low stand, id: 6
high stand, id: 7
zero torque, id: 8
wave hand1, id: 9
wave hand2, id: 10
shake hand, id: 11
Lie2StandUp, id: 12
Updated Test Option: Name = None, ID = None
Enter id or name:
9
Test: wave hand1, test_id: 9
Updated Test Option: Name = wave hand1, ID = 9
[ClientStub] send request error. id: 927663152790495
```
I would like to know how to use it in the mujoco simulator.

thomasschichl, 2025-05-15:
Did you find a solution for the [ClientStub] error?- none of the high_level example is working in mujoco for me.

jfelixre, 2025-08-25:
I also have the clientstub error