unitreerobotics/unitree_sdk2_python #65: 请提供audio client播放音频的python接口

JunweiLiang, 2025-04-26:
我看在C++版本里已经有播放wav文件的接口了：[这里](https://github.com/unitreerobotics/unitree_sdk2/blob/main/example/g1/audio/g1_audio_client_example.cpp#L104)
请添加对应的python接口，谢谢！目前只有少量的[接口](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/unitree_sdk2py/g1/audio/g1_audio_client.py)

Fatyhich, 2025-05-02:
Just translated version to save time fore somebody:

# Please provide Python interface for audio client to play audio #65

- I see that the C++ version already has an interface for playing WAV files: [here](https://github.com/unitreerobotics/unitree_sdk2/blob/main/example/g1/audio/g1_audio_client_example.cpp#L104) Please add the corresponding Python interface, thank you! Currently there are only a few interfaces available [here](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/unitree_sdk2py/g1/audio/g1_audio_client.py)

lvhaidong, 2025-05-17:
上面的播放、暂停我全部打通了。到时可以来找我要代码。 合并一下太慢了。 只需要传入一个wav文件，我会返回pcm_data，到时直接调用播放即可，停止我也在本地测试通过了。