unitreerobotics/unitree_sdk2_python #73: ImportError: Circular import on fresh installation

rayrsys, 2025-05-29:
Hello,

I am encountering a persistent circular import error when trying to use `unitree_sdk2py`, even in a completely clean environment.

**System Information:**
* OS: Ubuntu
* Python: Tried with both Python 3.12 and Python 3.8 (using fresh Conda environments)

**Problem:**
When importing any part of the library, it fails with the following traceback:
Traceback (most recent call last):
File "&lt;stdin>", line 1, in &lt;module>
File "/path/to/your/env/lib/python3.8/site-packages/unitree_sdk2py/init.py", line 1, in &lt;module>
from . import idl, utils, core, rpc, go2, b2
ImportError: cannot import name 'b2' from partially initialized module 'unitree_sdk2py' (most likely due to a circular import) (/path/to/your/env/lib/python3.8/site-packages/unitree_sdk2py/init.py)

**Steps to Reproduce:**
This error can be reproduced reliably with the following steps in a clean Conda environment:

1.  `conda create -n bug_report_env python=3.8`
2.  `conda activate bug_report_env`
3.  `pip install -U pip`
4.  `git clone https://github.com/unitreerobotics/unitree_sdk2_python.git`
5.  `cd unitree_sdk2_python`
6.  `pip install .`
7.  `python -c "from unitree_sdk2py.core.channel import ChannelFactoryInitialize"`

This consistently results in the `ImportError`. I have also tried installing via `pip install unitree_sdk2py==1.0.1` directly, and the result is the same. The error seems to be in the package's structure itself.

Thank you for your help.

rayrsys, 2025-06-08:
i havent been able to find it

lbypa, 2025-06-25:
Use ``pip install -e <repo_location>`` instead.

LBYPatrick, 2025-06-25:
Use ``pip install -e <repo_path>`` instead.

Kaweees, 2025-12-14:
The following merge requests fix this:
#126 
#107