unitreerobotics/unitree_sdk2_python #82: [Unitree_SDK2 (cpp)] Failed to initialize ROS node or ChanelFactory Init

theanhnguyen3112, 2025-06-24:
In my C++ code file, when I initialize both the ROS 2 node and the Unitree ChannelFactory, there is often a conflict. The code builds successfully, but when I run it with ros2 run, it throws an error: "failed to create domain, error: Precondition Not Met".

A simple workaround is to change the ROS domain ID before running, but since I'm developing multiple packages simultaneously, I’d prefer not to use that solution.

Is there any other way to resolve this?

comeonbabydontbeshy, 2025-06-24:
Have you try init ros2 node first and then init the Unitree ChannelFactory?

SURE3187774683, 2025-06-30:
I encountered the same problem when using Python code and would like to know if there is a solution.

theanhnguyen3112, 2025-06-30:
I encountered this problem with Cpp code. I not encountered this problem when using Python code with newest commit. 
With Cpp, before run code, you can change cycloneDDS to fastDDS.

SURE3187774683, 2025-06-30:
Thank you for your answer. I didn't encounter this problem when running it on my own laptop with Python, but I faced this issue when running it on the robot's PC2. It is very likely that the DDC selection is incorrect. Thank you.

Nikita-Dudorov, 2025-06-30:
facing same issue trying to run control on go2w internal PC, no problem while using external PC

flyhuiyi, 2025-08-10:
I also encountered the same initialization conflict issue while running my Python code, with error messages：
[ChannelFactory] create domain error. msg: Occurred upon initialisation of a cyclonedds.domain.Domain