unitreerobotics/unitree_sdk2_python #88: Segfault after sending a command from the Go2 dog

egormalyutin, 2025-07-11:
Hello. I'm trying to run examples on dog's Jetson. I've successfully installed the SDK, but when I'm running any of the examples I'm getting a segfault after sending a message. For example:

```
> python example/helloworld/subscriber.py
Subscribe success. msg: UserData(string_data='Hello world', float_data=1752209916.486942)
```

```
> python example/helloworld/publisher.py
Segmentation fault (core dumped)
```

Looks like the message gets sent successfully (for example `go2_sport_client.py` kind of works but still segfaults after sending a command), but for some reasons client can't receive the reply.

Zhang-Yongqi, 2025-09-07:
I also encountered this problem. Have you solved it?

egormalyutin, 2025-09-07:
@Zhang-Yongqi No. Seems like cyclone dds being built from sources is the problem, but I have no idea what the problem is exactly. However the binaries shipped in https://github.com/unitreerobotics/unitree_sdk2 do work and I ended up using c++ anyway. You can also just write your own bindings if you want to

egormalyutin, 2025-09-07:
@Zhang-Yongqi ok but I highly suggest to not waste your much of your time trying to make it work. I've spent an embarrassing amount of time on it while all these issues could be solved by just writing custom bindings 😅