unitreerobotics/unitree_sdk2_python #94: Integration of External Camera with Browser Streaming + Photo Capture at LiDAR-Mapped Waypoints on Go2 EDU

robotsforhumanity, 2025-07-31:
We want to use the Unitree Go2 EDU robot to map an environment using its onboard LiDAR, generating a series of waypoints along the infrastructure. The idea is for the robot to navigate through these points, and each time it reaches a waypoint, it should send a signal from PC1 (the robot’s onboard computer) to an external PC2 connected to a camera. When PC2 receives the signal, it should trigger the camera to take a photo. Additionally, the camera must be streaming live video through a web browser so users can access the feed in real time.

Main Goal:
	•	Enable autonomous navigation on the Go2 EDU using LiDAR.
	•	Send a signal from the robot (PC1) to a second computer (PC2) whenever a waypoint is reached.
	•	Make PC2 capture a photo with the camera upon receiving the signal.
	•	Keep a live video stream from that camera available through a browser.

Main Technical Question:
We’re unsure if it’s possible to access or modify PC1 on the Go2 EDU to implement this logic.

ahsan155, 2026-01-06:
@robotsforhumanity Did you make any progress in your project? I am going to have access into Unitree G1 robot soon. I am reading the documentation. I would like to know what localization capability the sdk has. The documentation says that it calls on some SLAM and navigation interface. So, is the sdk going to make api call to some online localization service developed by unitree? Can this service pinpoint robot location on pcd map?

makephysical-netizen, 2026-11-03:
Hi, I’m Alex from Benewake.

For your Go2 inspection route, I’d first repeat the same photo stop with the onboard LiDAR and check position and heading against a fixed reference. If that misses your capture tolerance, SR18 could be evaluated as an external 3D mapping sensor: 18 channels over 360° × 48°, at 190 g. Its RS485 link runs at 3.125 Mbit/s, so host-interface and SLAM integration would need checking; I haven’t verified it on Go2.

What position and heading tolerance does each photo require?